virtualabs/btlejack #3: Feature-Request BLE-Relay

savek-cc, 2018-08-13:
Hi! Have you thought about adding a BLE-Relay feature? (Run two laptops, each equipped with a Micro:Bit and relay/hijack a connection by forwarding packets (via IP) between the two machines)

virtualabs, 2018-08-13:
Yes, I did. I think this can be achieved by Nordic's SoftDevice alone on any nRF51822, but we've already created  [Btlejuice](https://github.com/digitalSecurity/btlejuice) for this purpose. You may also have a look at [GATTacker](https://github.com/securing/gattacker), written by Slawomir Jasek.

But I'm afraid, this feature will not be part of this project.

closed 2018-08-13 by virtualabs

savek-cc, 2018-08-13:
Btlejuice requries the client to actively connect to it. I am thinking about a pure range extender/relay station attack that does work on link-layer. I.e. allows you to extend/record a BLE connection between two devices that was established via LE Secure Connections in the past.
As you are able to read and write link-layer packets, I think it should be possible to extend this project with such a functionality?