virtualabs/btlejack #5: Fix for this issue

Mr-iX, 2018-08-13:
Is it possible to fix this issue with a software update or requires this new hardware?

If a software update from the manufacturers is enough, must both connected devices be patched?

virtualabs, 2018-08-13:
> Is it possible to fix this issue with a software update or requires this new hardware?
Which issue ?

The python package embeds the firmware and the `-i` option must be used to install the latest firmware version on all of your Micro:Bit devices.
If you plan to use multiple Micro:Bit, yes you need to upgrade all of them. Btlejack will not stop working, but you may experience issues if your Micro:Bit devices do not run an up-to-date version of our firmware.

Mr-iX, 2018-08-13:
Sorry for being inaccurate. I‘ve meant if Apple/Google or other companies could fix the Bluetooth LE security issue for their devices, or if a security fix is only applicable for new devices, because new hardware is needed?

virtualabs, 2018-08-13:
I see. It is mostly a software issue as application developers shall ensure data integrity and authenticity, or use BLE secure connections and let the protocol enforce both of them (even if the attack works, an attacker would not be able to perform unauthorized actions on the hijacked device).

Goole or Apple could not do anything to avoid third-party application using unsecured BLE connections, I'm afraid. Bluetooth SIG may issue a fix in a new version of the Bluetooth specifications, and chip makers like Nordic Semiconductor, TI or NXP will have to update their SDKs and BLE stacks to include this fix.

Mr-iX, 2018-08-14:
But the jamming will be always possible, won’t it? Only the authenticity can be verified, but the connection itself cannot be garuanteed, right?

virtualabs, 2018-08-15:
Well, Bluetooth 5 specifies a brand new channel hopping algorithm (not yet defeated), called CSA #2. If an attacker tries to hijack an active connection that uses this algorithm, it would not be able to follow it and jam it. If an attacker monitored a connection from the start, well, this connection cannot be guarantied. Unless someone finds a way to break this new CSA#2 algorithm. Jamming has always be a threat for wireless communications, and it is almost impossible to avoid it (as far as I know).

closed 2018-08-15 by Mr-iX