virtualabs/btlejack #7: recommending using sudo pip to install

lindsayemarc, 2018-08-15:
We should be recommending 

pip3 install XXXXX --user

this will install the package for the current user

README.md needs to be updated

Sontras, 2018-08-20:
On macOS Sierra (10.12.6) ```# pip3 install btlejack --user```  gave me a warning.
```
  The script btlejack is installed in '/Users/Sontras/Library/Python/3.7/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
```

```# sudo -H pip3 install btlejack``` did the trick.
I cannot tell if this is macOS Sierra specific.

lindsayemarc, 2018-08-23:
When you run pip with sudo, you run setup.py with sudo. In other words, you run arbitrary Python code from the Internet as root. If someone puts up malicious code in btlejack (or any btlejack dependency) and you install it, you give an attacker root access to your machine. 

@Sontras I did not receive that warning. A quick google search of your issue shows that is pretty common though. This gh issue seems particularly interesting - https://github.com/pypa/pip/issues/5316

maybe you have a trailing slash in your path?

closed 2022-11-18 by virtualabs