virtualabs/btlejack #12: ESP32

JoanMCD, 2018-08-17:
From your point of view would it possible to port your firmware to the ESP32 ?
BR

mstrmnn, 2018-08-17:
I'm yearning for a ESP32 solution, too. It's so much cheaper!

virtualabs, 2018-08-17:
ESP32 is a whole different architecture, and I'm not sure it would be able to:
- sniff BLE packets from raw demod (I'm actually using the same sort of  hack Travis Goodspeed did with the nRF24L01)
- allow receiving BLE packets with CRC check disabled

Moreover, a different architecture means a different code, and of course a different BLE stack I have to develop. 

That's huge work, really. I've almost spent 1.5 year developing btlejack, and don't think if I'm ready to dig into a new architecture and port btlejack to it. Yes, it would be cheaper, but would take quite some time to achieve. Since I'm no ESP32 expert, I'm not able to tell you how much it would take to do so.

If you want cheaper hardware to use btlejack, I'm currently working with @slawekja to support all these cheap chinese nRF51822 modules out there ($3 each).

virtualabs, 2018-08-27:
BLE400 and Adafruit's Bluefruit LE sniffer are now supported by Btlejack.

closed 2018-08-27 by virtualabs

virtualabs, 2018-08-27:
I suppose this module UART port is directly wired to a microcontroller not directly to USB, IDK if it will work with this setup.

browntownington, 2023-08-03:
Hey @SlawekJA did you have any tips to get these Chinese nRF51822 working? I've seen people ask @virtualabs but not sure he's monitoring the issues atm.