virtualabs/btlejack #13: UnicodeDecodeError README.rst

Sontras, 2018-08-18:
macOS 10.12.6

```
$ pip install btlejack --user
Collecting btlejack
  Using cached https://files.pythonhosted.org/packages/1e/7a/73e21237fb5b1dbfc7c9982d45d8469f086ec0a4f921f93c9f6149915c33/btlejack-1.1.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/b3/8p1bm4r536l454jvxngg8rrr0000gn/T/pip-install-4p0o_ezu/btlejack/setup.py", line 25, in <module>
        long_description = read('README.rst'),
      File "/private/var/folders/b3/8p1bm4r536l454jvxngg8rrr0000gn/T/pip-install-4p0o_ezu/btlejack/setup.py", line 9, in read
        return open(os.path.join(os.path.dirname(__file__), fname)).read()
      File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 1523: ordinal not in range(128)

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/b3/8p1bm4r536l454jvxngg8rrr0000gn/T/pip-install-4p0o_ezu/btlejack/

```
If I convert README.rst to  ASCII with ```iconv -c -f UTF-8 -t ascii README.rst.bak > README.rst``` everything works fine.

Sontras, 2018-08-19:
Hello and thanks for the reply,

my MacPort python installation is also bit messed up.

Thanks

Sontras, 2018-08-20:
Just in case someone is running into the same problems with macOS Sierra:
You have to have Xcode installed.
You have to have ```/usr/local/bin``` in your PATH variable.

**Install Xcode command line tools and homebrew**
```
# xcode-select --install
# ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
```
**Install python3**
```
# brew install python
```
**Install btlejack**
```
# sudo -H pip3 install btlejack
```

Cheers

closed 2018-08-20 by Sontras

floyd-fuh, 2018-08-21:
Actually, looking at this error message, the OP was using Python 3, but on MacOS you might not have your console set up as UTF-8 (in my experience there are some funny python 2 libraries that won't work anymore if you do that). So the easier fix for this problem is:

```
export LC_CTYPE=utf-8
export PYTHONIOENCODING=utf-8
```

At least that worked for me