virtualabs/btlejack #16: Jamming "normal" Bluetooth devices

JsBergbau, 2018-08-22:
Is it possible to just jam normale Bluetooth devices, e.g. Bluetooth Speakers? It would be so great to just relax and have silence without these kids playing their loud music everywhere?

jlusiardi, 2018-08-22:
i understand the issue ;) but as far as i understood it is bluetooth low energy only. perhaps some of those speakers use this technology ...

virtualabs, 2018-08-22:
Btlejack does not support Bluetooth Classic, I'm afraid. It would be useful to make these Bluetooth speakers shut (or far better make them play other songs), but this cannot be achieved with Btlejack.

closed 2018-08-22 by virtualabs

JsBergbau, 2018-08-23:
Thanks for clarifying so fast. Do you know any other cheap solution to achive jamming these connections?

virtualabs, 2018-08-23:
Bluetooth uses a channel hopping mechanism and 79 channels of 1MHz each, so it is rather difficult to jam unless you are not synchronized with a specific connection or jamming all the channels at the same time. It is not easy at all.