virtualabs/btlejack #21: SWD programmer Vs BLE400

minkione, 2018-09-04:
Hi,
I was planning to play with a waveshare BLE4000 board and the walkthrough to the wiki.
Do you have some external link about programming the board with the SWD Programmer?

So far I have found this resource, but is not clear how to push the new firmware...
https://rlangoy.github.io/OpenOCD_MBED_FRONTEND/HWExample.htm

bogiton, 2018-10-19:
This is probably an issue for the btlejack-firmware, but I am also struggling to get this flashed into the BLE400 using RPi/OpenOCD.
I seem to have successfully attached the RPi to the BLE400 using SWD, as described in the link you provided, and I have also established a working OpenOCD session, where I issue the following commands:
```
init
reset
halt
nrf51 mass_erase
program /<path-to-the-firmware>/ble400.hex verify
reset
shutdown
```
Everything looks ok in the end, apart from a not so bright lit LED2.
Anyhow, I then run `btlejack -d /dev/ttyUSB0 -s` and stays there forever. No errors, no nothing.
Is this a proper way to flash the firmware to the BLE400 or there is something missing? Also, is there any way to verify that the flashing was successful?

bogiton, 2018-10-25:
Just an update to this. I ended up recompiling the latest firmware for BLE400 and then flashed it, following the exact aforementioned steps and now it is working fine.
@minkione Make sure you have the correct swd pins defined in interface/raspberrypi2-native.cfg for openocd to see the core51822/ble400. I had to add this line: `bcm2835gpio_swd_nums 25 24`

virtualabs, 2018-11-27:
Thank you for this information. It may be interesting to add it to Btlejack's documentation.

closed 2018-11-27 by virtualabs

bogiton, 2018-11-28:
@virtualabs Thank you for this epic tool :)

avageo, 2020-10-04:
**@bogiton**  Can you please give some tips on how to recompile the BLE400 firmware? I am a complete beginner and would like to try BtleJack with BLE400. Unfortunately I get a message after starting the tool that I have to update the firmware.

Thank you in advance!