virtualabs/btlejack #25: pcap.py: ll_phdr: Index out of range

mprt, 2018-09-25:
# btlejack -c any -o test -x ll_phdr 
BtleJack version 1.3

[i] Detected sniffers:
 > Sniffer #0: version 1.3
 > Sniffer #1: version 1.3
 > Sniffer #2: version 1.3
LL Data: 05 22 79 b9 fa 13 ac 42 18 49 52 ee f3 0c 2d ba 9a af e1 96 45 03 02 00 18 00 00 00 48 00 ff ff ff ff 1f 0f
[i] Got CONNECT_REQ packet from yy:yy:yy:yy:yy:yy to xx:xx:xx:xx:xx:xx
 |-- Access Address: 0xaf9aba2d
 |-- CRC Init value: 0x4596e1
 |-- Hop interval: 24
 |-- Hop increment: 15
 |-- Channel Map: 1fffffffff
 |-- Timeout: 720 ms

LL Data: 03 06 0c 07 0f 00 0d 41
LL Data: 0b 06 0c 08 5a 00 00 12
LL Data: 03 17 03 00 00 00 00 00 00 00 00 00 00 94 33 fb b2 d8 9d 6a 99 8e ac 99 39
LL Data: 0b 0d 04 8b 21 fc c2 6d 1a 1c aa 96 49 d4 f7
Traceback (most recent call last):
  File "/usr/local/bin/btlejack", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/btlejack/__init__.py", line 329, in main
    supervisor.process_packets()
  File "/usr/local/lib/python3.6/dist-packages/btlejack/supervisors.py", line 432, in process_packets
    self.on_packet_received(pkt)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/supervisors.py", line 476, in on_packet_received
    self.on_ll_packet(packet)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/ui.py", line 409, in on_ll_packet
    self.output.write_packet(ts_sec, ts_usec, self.access_address, packet.data)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/pcap.py", line 70, in write_packet
    payload = self.payload(aa, packet)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/pcap.py", line 96, in payload
    packet[2],
IndexError: index out of range

# btlejack -c xx:xx:xx:xx:xx:xx -o test -x ll_phdr 
BtleJack version 1.3

[i] Detected sniffers:
 > Sniffer #0: version 1.3
 > Sniffer #1: version 1.3
 > Sniffer #2: version 1.3
LL Data: 05 22 79 b9 fa 13 ac 42 18 49 52 ee f3 0c aa bb 9a af f3 d0 ac 03 07 00 18 00 00 00 48 00 ff ff ff ff 1f 08
[i] Got CONNECT_REQ packet from yy:yy:yy:yy:yy:yy to xx:xx:xx:xx:xx:xx
 |-- Access Address: 0xaf9abbaa
 |-- CRC Init value: 0xacd0f3
 |-- Hop interval: 24
 |-- Hop increment: 8
 |-- Channel Map: 1fffffffff
 |-- Timeout: 720 ms

LL Data: aa 2b 4b 84 23 6b 73 ee 70 41 f3 70 2b c1 6b 4c e7 59 1c c3 9c 66 fd ed 9d 9c 0c 9c e4 f0 66 85 6e a5 e9 6a c1 cc 6d 5a 56 f6 9b 4c c2
LL Data: 16 1f 31 cf 51 68 ad d6 8e f0 ba 2f 29 2e e8 3e 8d 1f 00 85 0a c5 80 65 cc a8 e2 90 56 b6 b9 25 6f
LL Data: 0a 1f 93 da 18 b6 e4 a2 da 38 f8 85 10 a9 69 c3 23 f3 c1 55 7b da e8 13 86 13 13 37 46 9b d7 cd 50
LL Data: 06 1f 62 2b 5a 17 9c 7a 88 86 70 b3 c4 3f 9e a0 4d f4 c0 cc dc bd fc 62 9f 51 9d 38 ad 27 fd d4 97
LL Data: 02 0f 9e 9b 0d 3e 43 ba 80 2a 20 60 ea 84 1c 69 d7
LL Data: 0a 15 5e 9d f4 ed 59 2a 16 37 73 0d 5f 11 57 3a 6e 99 06 e5 9b 20 1a
Traceback (most recent call last):
  File "/usr/local/bin/btlejack", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/btlejack/__init__.py", line 329, in main
    supervisor.process_packets()
  File "/usr/local/lib/python3.6/dist-packages/btlejack/supervisors.py", line 432, in process_packets
    self.on_packet_received(pkt)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/supervisors.py", line 476, in on_packet_received
    self.on_ll_packet(packet)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/ui.py", line 409, in on_ll_packet
    self.output.write_packet(ts_sec, ts_usec, self.access_address, packet.data)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/pcap.py", line 70, in write_packet
    payload = self.payload(aa, packet)
  File "/usr/local/lib/python3.6/dist-packages/btlejack/pcap.py", line 96, in payload
    packet[2],
IndexError: index out of range`

virtualabs, 2018-09-25:
Ok, I need to reproduce this. Is the connection encrypted (does the device needs pairing) ?

mprt, 2018-09-26:
I'm not sure what exactly the MicroBits received, but the hcilogs tell me that there's some unencrypted GATT traffic going on and then they pair using LE Secure Connections JustWorks.

mprt, 2018-11-27:
couldn't reproduce.
Got a new version of the App in question and it looks like they fixed whatever went wrong...

closed 2018-11-27 by mprt