virtualabs/btlejack #30: computing CRCInit value

Wini3114, 2019-01-10:
When I try to follow a connection, after executed -s command, it always stops at computing CRC init value and computes without an ending, but doesn't find anything. Is there any trick or suggestion on what to do?
And what exactly is the CRCInit value?
Thanks in advance!

thisguy23, 2019-01-17:
same here.
[image]

virtualabs, 2019-01-18:
Allright, this one puzzled me a bit but I think I know what is happening.

When Btlejack sniffs for available access addresses, it relies on the nRF51822 sniffing mode which is basically a hack (this chip has no "official" sniffing capability). That said, the sniffed access addresses are prone to errors and some bits of the actual access address may have been sniffed wrong, leading to an inexisting access address value.

If you try to follow the connection corresponding to this access address, you will fail as it does not correspond to any existing connection and therefore Btlejack won't be able to capture packets and recover this connection's CRC initial value (CRCInit). If you want more information about this value, you might consider reading the Bluetooth 4.2 core specifications or even watch my talk at DEF CON or Mike Ryan's talk at WOOT13 (the man who started it all :p). 

Anyway, when using Btlejack -s option, please consider a valid access address when more than 5 packets has been caught. Try again with that in mind, and come back here if it still does not work.

thisguy23, 2019-01-18:
I have found that forcing more packets will allow it to complete, but then it will stall again on "computing hop increments" 
Thanks for all your efforts, and assistance.