virtualabs/btlejack #34: Add some installation hints for Raspberry Pi (esp. Zero W)

mh-, 2019-02-08:
When I tried to install btlejack on a Raspberry Pi Zero W, I followed the instructions from this [Blog post: SSH into Pi Zero over USB](https://desertbot.io/blog/ssh-into-pi-zero-over-usb).
This enables virtual ethernet over USB and allows to initially connect to the Pi Zero W, to set it up and install stuff like btlejack.
In order to use btlejack, I needed the USB port for the sniffers, and therefore configured the Pi Zero W to connect to WLAN, but I forgot to disable the virtual ethernet driver, i.e. I didn't remove `dtoverlay=dwc2` and `modules-load=dwc2,g_ether`.
This basically broke btlejack completely, because this driver tried to use the sniffer USB connection, and all kinds of strange issues occurred: The Pi Zero W became completely unresponsive on ssh; when I tried this on a Pi 3 B+, the usb devices /dev/ttyACM0, /dev/ttyACM1, ... were blocked and couldn't be accessed by btlejack; and many more strange issues.

So I suggest adding a hint for Raspberry Pi users:

> ### Hint for using btlejack on a Raspberry Pi:
> If you have previously enabled **virtual ethernet over USB** (RNDIS), e.g. to setup a Raspberry Pi Zero W over USB, you need to disable this again (i.e. remove `dtoverlay=dwc2` from boot/config.txt and `modules-load=dwc2,g_ether` from boot/cmdline.txt, then `sudo reboot`), because this would otherwise interfere with the sniffers' USB connections.

virtualabs, 2019-02-09:
You are absolutely right, this could help some users trying to make it work on a Raspberry Pi. I did not try it myself, so I guess it will be improved by contributors (me included when I will get my hands on a Raspberry Pi ...). 

Can you do a pull request with the modified readme file ? Or if you do not want to, I will modify it accordingly.

virtualabs, 2019-02-11:
Allright, thank you again for your contribution !

closed 2019-02-11 by virtualabs

chr1sdoofer, 2023-04-08:
Sorry for chiming in this late, I've recently learned about this project and would love to set it up on my Pi Zero W.
Unfortunately I am not able to get it running with the hints provided. Since it's been 4 years by now, are those hints still up-to-date? I haven't had any of those lines in my config.txt & cmdline.txt and when executing `btlejack -i` I get the following error:

`[i] No sniffer found, make sure all your devices are mounted as mass storage devices before flashing.`

Sadly I haven't found any other resources on btlejack and Raspberry Pi's besides this thread, which is why I decided to post on here. If there are any resources available, I'd love to know!