virtualabs/btlejack #40: Multiple Adafruit Bluefruit LE Sniffer needed?

LaborEtArs, 2019-04-24:
Sorry, no issue, but a usage question...
It is recommended to use three BBC Micro:Bit devices to 'spy' on all three advertising channels simultanously.
Would I need to use three Adafruit Bluefruit LE Sniffers as well?

chreck, 2019-09-23:
Maybe you got the answer already. I could not let it run on a macOS with btlejack. But if you are interested in using it with wireshark you only need one of these Bluefruit sniffer. Because Bluefruit can listen to 3 channels at once. You can setup it with this article https://learn.adafruit.com/introducing-the-adafruit-bluefruit-le-sniffer

virtualabs, 2019-09-24:
Bluefruit can listen to 3 channels successively, as far as I know, not simultaneously. Using 3 Micro:Bit devices allow simultaneous sniffing on the 3 advertising channels.

LaborEtArs, 2019-09-25:
Many thanks for your answers! However, meanwhile I dropped this project :-(

closed 2019-09-25 by LaborEtArs