virtualabs/btlejack #59: Passive Radiator?

ArcadeHustle, 2019-12-30:
Can this code be used to passively retransmit packets it sees? I would like to test some passive entry systems that rely on the proximity of the beacons. I'd like to capture them in a remote location and real time replay them (over ethernet) to a different location for example.

RCayre, 2020-08-11:
If you want to replay advertisements, btlejack is not able to sniff or transmit advertisements yet. However, the hardware supports it, so you can easily add these features if you modify the code of the firmware.

virtualabs, 2020-08-31:
RCayre is right. Btlejack does not handle this use case, as it is not able to transmit easily an arbitrary packet.

closed 2020-08-31 by virtualabs