virtualabs/btlejack #61: Characteristics Discovery does not always Work

mindfuckup, 2020-02-24:
Hi

**Issue**
The `discover` command in a hijacked connection does not always work.

**Steps to reproduce**
Hijack a connection:
```
# btlejack -z -t -f 0x50656a9a
BtleJack version 2.0

[i] Stored connections cleared
[i] Detected sniffers:
 > Sniffer #0: fw version 2.0

[i] Synchronizing with connection 0x50656a9a ...
✓ CRCInit = 0x23297f
✓ Channel Map = 0x1fff0001ff
✓ Hop interval = 9
✓ Hop increment = 5
[i] Synchronized, hijacking in progress ...
[i] Connection successfully hijacked, it is all yours \o/
>> 16 09 05 00 04 00 1b 4b 00 63 00
>> 1a 09 05 00 04 00 1b 4b 00 64 00
>> 16 09 05 00 04 00 1b 4b 00 63 00
>> 1a 09 05 00 04 00 1b 4b 00 63 00
```
List characteristics shows a error instead of the characteristics:
```
>> 06 09 05 00 04 00 1b 4b 00 4c 00
btlejack> Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/__init__.py", line 355, in main
    supervisor.process_packets()
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/supervisors.py", line 38, in process_packets
    self.on_packet_received(pkt)
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/ui.py", line 663, in on_packet_received
    super().on_packet_received(packet)
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/supervisors.py", line 331, in on_packet_received
    self.on_ll_packet(packet)
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/ui.py", line 628, in on_ll_packet
    self._pt.on_ll_packet(packet)
  File "/usr/lib/python3.8/site-packages/btlejack-2.0.0-py3.8.egg/btlejack/ui.py", line 274, in on_ll_packet
    response = L2CAP.from_bytes(packet.data[12:]).payload.payload
AttributeError: 'NoneType' object has no attribute 'payload'
```

I get this error nearly everytime I use it. I rarely get the desired output.

Thanks & best regards,
Emanuel

virtualabs, 2020-02-24:
Well, it could be due to an encrypted connection or a bug in btlejack. Does your BLE device require some kind of pairing ?

mindfuckup, 2020-02-25:
No, the device does not pair at all. I'm able to sniff new connections and also use the write command in hijacked connections. Only the discover command does not work.