virtualabs/btlejack #62: multiple ble400 boards

coelner, 2020-02-28:
Did someone test this with multiple devices? I tried this but only the first device get used.

```btlejack -d /dev/ttyUSB1 -d /dev/ttyUSB0 -d /dev/ttyUSB2 -s
BtleJack version 2.0

[i] Enumerating existing connections ...
^C[i] Quitting
```
And only one of the uart adapters is powered up and show traffic. It is not a power problem.

virtualabs, 2020-03-29:
I've never tested btlejack with multiple ble400 boards, but never had any issue with microbits. It is supposed to run the same firmware with minor differences (UART configuration differs), so I cannot see why it would not behave the same. I'll reach some friends of mine who have a few BLE400 boards available and see if they get the same behavior. What version of btlejack are you using ?

coelner, 2020-03-30:
Version 2 or did you meant the firmware version? I tried it with the select() pullrequest and master branch.

coelner, 2020-08-14:
@virtualabs Could you make any step forward?

coelner, 2021-04-08:
https://github.com/virtualabs/btlejack/blob/7cd784a5c9ead820bdc1b1745d2d8e0773e6facb/btlejack/supervisors.py#L85-L88
this is intentionally , but why?

closed 2021-04-08 by coelner