virtualabs/btlejack #71: Reading Source/Destination MAC Address

vpetersson, 2020-10-13:
Hi guys,

I've started to dabble with BLE in the last year or so as I've built out [Sonar](https://github.com/databat-io/sonar). The goal is to do people analytics passively using BLE/BT on a Raspberry Pi. The current version is using the built-in BT chipto sniff Advertising data. However, I'm looking to take it a step further by sniffing BLE using a Microbit or Adafruit BLE sniffer. 

I've got the hardware already and started to play with it. However, after analyzing the PCAP file, I noticed that the access Address in the BLE Link Layer isn't what I expected it to be (i.e. a MAC Address):

[image]

Now, before I go further down the rabbit hole, I just wanted to sanity check and see if it is indeed possible to capture the MAC Address(es) using btlejack, or if I'm running up against some kind of encryption issues.

virtualabs, 2020-10-22:
Hi, 

The fact is Access Address is not Bluetooth Address (somehow a MAC address). Bluetooth addresses are only used in advertisement packets and connection requests, but once a connection established these values are no more used. The link created between two devices is identified by an Access Address, a unique 32-bit value that has absolutely no relation to the devices Bluetooth addresses. This Access Address is then used to identify the packets sent back and forth between both devices, along with other specific values (CRCInit, etc...). 

Hope this helps :)

vpetersson, 2020-10-22:
Thanks, @virtualabs. Yes, much appreciated. Just one more question (i hope :) -- is the Access Address static for a given device, or is it randomly generated over time?

virtualabs, 2020-10-22:
It is randomly generated by a device initiating a connection, but not all bits are random. A set of specific rules must be followed to generate a valid Access Address (see BLE specs for more details).

vpetersson, 2020-10-29:
Thanks, @virtualabs! Closing this out.

closed 2020-10-29 by vpetersson