virtualabs/btlejack #72: No Results with RPI4 running from SSD (USB -> SSD Bridge)

realhagie, 2020-10-21:
Trying to sniff BLE connections (tons are in the air here - like MI Band, Samsung TV's Solar Charge Controller) I get no Results from my Micro:Bit Device. I flashed the 2.0 FW and updated the Software to latest GIT. 

```root@raspberrypi:~# btlejack -s
BtleJack version 2.0

[i] Enumerating existing connections ...
```
or 
```
root@raspberrypi:~# btlejack -c any
BtleJack version 2.0

[i] Detected sniffers:
 > Sniffer #0: version 2.0
```
Stays forever.

Device Information: 

```
root@raspberrypi:/mnt/MICROBIT# more DETAILS.TXT
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9901000052964e45006b0011000000470000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0253
Bootloader Version: 0243
Git SHA: 64359f5c786363065a41ec15c348e3d53568da03
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0x53375800
Remount count: 0
URL: https://microbit.org/device/?id=9901&v=0253
```

Could this be a problem with the attached USB (SSD) or the RPI 4 ?

Thanks

realhagie, 2020-10-21:
I just installed a fresh RPI3 on a standard sdcard and have the same (none) results .. Here are the system Messages 👍 
``` 
Oct 21 21:38:22 raspberrypi kernel: [ 2061.389146] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
Oct 21 21:38:22 raspberrypi kernel: [ 2061.522916] usb 1-1.2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice=10.00
Oct 21 21:38:22 raspberrypi kernel: [ 2061.522932] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 21 21:38:22 raspberrypi kernel: [ 2061.522945] usb 1-1.2: Product: DAPLink CMSIS-DAP
Oct 21 21:38:22 raspberrypi kernel: [ 2061.522956] usb 1-1.2: Manufacturer: ARM
Oct 21 21:38:22 raspberrypi kernel: [ 2061.522968] usb 1-1.2: SerialNumber: 9900000052964e45006b0011000000470000000097969901
Oct 21 21:38:22 raspberrypi kernel: [ 2061.524467] usb-storage 1-1.2:1.0: USB Mass Storage device detected
Oct 21 21:38:22 raspberrypi kernel: [ 2061.530294] scsi host1: usb-storage 1-1.2:1.0
Oct 21 21:38:22 raspberrypi kernel: [ 2061.534701] hid-generic 0003:0D28:0204.0002: hiddev96,hidraw0: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-3f980000.usb-1.2/input3
Oct 21 21:38:22 raspberrypi kernel: [ 2061.538329] cdc_acm 1-1.2:1.1: ttyACM0: USB ACM device
Oct 21 21:38:22 raspberrypi mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Oct 21 21:38:22 raspberrypi mtp-probe: bus: 1, device: 6 was not an MTP device
Oct 21 21:38:22 raspberrypi mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Oct 21 21:38:22 raspberrypi mtp-probe: bus: 1, device: 6 was not an MTP device
Oct 21 21:38:23 raspberrypi kernel: [ 2062.570376] scsi 1:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
Oct 21 21:38:23 raspberrypi kernel: [ 2062.571138] sd 1:0:0:0: Attached scsi generic sg0 type 0
Oct 21 21:38:23 raspberrypi kernel: [ 2062.571967] sd 1:0:0:0: [sdb] 16512 512-byte logical blocks: (8.45 MB/8.06 MiB)
Oct 21 21:38:23 raspberrypi kernel: [ 2062.572471] sd 1:0:0:0: [sdb] Write Protect is off
Oct 21 21:38:23 raspberrypi kernel: [ 2062.664804] sd 1:0:0:0: [sdb] Attached SCSI removable disk

```
maybe it helps

realhagie, 2020-10-22:
Additional - I was not able to flash the device without generationg a "Fail.txt" on the mass storage. I moved the device over to my window machine - updated the latest Micro:Bit Firmware (via maintinance mode) and after that I used the normal mode to flash the btlejack fw 2.0 from the repo.

realhagie, 2020-10-22:
I tried Ubuntu 20 Desktop on my PC - I even downgraded the FW of the MicroBit down to 241 (from 253) everything without success. On Ubuntu 20 Desktop the MicroBit automatically mounts and installing the FW works without problems.
I flashed some demo hex on the MicroBit and checked to see if the pairing and advertising of the ble hardware is working and it does. I can query the MicroBit from my phone. Could there be some new hardware revision which don't allow low level access to the ble chip ?

virtualabs, 2020-10-22:
Allright, it looks like your BBC Micro:Bit is connected in DAPLink mode. This mode is a maintenance mode used to upgrade the firmware of its KL26 chip that is responsible of handling USB connections. It is normally enabled when pressing the reset button on the Micro:Bit while plugging it in, as described here https://os.mbed.com/daplink-on-kl26z/.

DAPLink may also be triggered if the firmware installed on the KL26 is in a wrong state, so I think you should try to update the DAPLink firmware first, and then try to upload Btlejack's firmware on it.

realhagie, 2020-10-22:
I did several times - each time I upgraded the DAPLink Firmware I used the Bootloader procedure (tried 241, 247, 250 and 253), after that the device rebootet and I used the .hex file and copied it over (on my windows box) or the last thing I tried was the ubuntu linux box where I used the -i parameter and the device bootet itself and the -c parameter found the sniffer V2
also the name of the device is MicroLink and not Maintenance. Everything looks fine so far. Just nothing detected.

realhagie, 2020-10-24:
Ok version I own is a V1.5 

```
micro:bit version | Board ID
-- | --
v1.3 | 9900
v1.5 | 9901
v2.0 | 9903 (reserved), 9904
```
could my problem source from a defective chip or may the revision 1.5 be incompatible with btlejack ?
On a sidenote will btlejack support the upcoming V2 with speaker and microphone ?

realhagie, 2020-10-26:
OK I just got another (v1.5) Version with the same results, will try now another OS like ubuntu instead of latest debian (rpi / pc).

realhagie, 2020-10-26:
Ok I tried now with ubuntu 20.04 LTS 32Bit on a RPI3 and have success.
Same Version on PC as 64 Version don't work. Could this some kind of missing dependency or is it a 64Bit Problem ??

```
root@ubuntu:~/btlejack# btlejack -c any
BtleJack version 2.0

[i] Detected sniffers:
 > Sniffer #0: version 2.0
LL Data: 05 22 29 64 b6 3a 0f 60 99 24 61 6d e3 94 cd c3 c7 43 e8 f1 e8 02 19 00 28 00 00 00 f4 01 ff ff ff ff 1f 08
[i] Got CONNECT_REQ packet from 60:0f:3a:b6:64:29 to 94:e3:6d:61:24:99
 |-- Access Address: 0x43c7c3cd
 |-- CRC Init value: 0xe8f1e8
 |-- Hop interval: 40
 |-- Hop increment: 8
 |-- Channel Map: 1fffffffff
 |-- Timeout: 5000 ms
```

Andy-45, 2021-03-13:
My new MicroBit V1.5 Board is running fine on RPi3B+ and on PC Kali Linux 

> Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
> Linux kali 5.10.0-kali4-amd64 #1 SMP Debian 5.10.19-1kali1 (2021-03-03) x86_64 GNU/Linux
> 
but I also could not flash via btlejack -i switch, but copy/paste on Win10 works.

I get good results with bluetooth devices in 2-3 meter range, my nRF dongle was detected immediatly. 
Same with the bluetooth keyboard.

Thanks for this good work, Damien!

closed 2022-11-18 by virtualabs