virtualabs/btlejack #80: micro:bit v 2.21

hanzorz, 2023-03-25:
Has anyone been able to successfully get the 2.21 version to sniff connections?

virtualabs, 2023-04-02:
Current implementation does not correctly detect access addresses due to the way Nordic's nRF52 works. This is a known limitation, but some folks gave me a few tips about how to fix that (using multiple reception pipes), but I still need to find some time to fix the firmware...

hanzorz, 2023-04-23:
I forgot to follow up with you on this. I ended up getting everything working on three Micro:Bit V2.1s and can confirm that all features work. The access addresses seem to be correct since I’m able to hijack connections and write to the slave devices.

closed 2023-04-23 by hanzorz

TechTune-1, 2023-10-14:
> I forgot to follow up with you on this. I ended up getting everything working on three Micro:Bit V2.1s and can confirm that all features work. The access addresses seem to be correct since I’m able to hijack connections and write to the slave devices.

Greetings, I would like to know how you’ve got everything working on three Micro:Bit V2.1s. Could you please explain what you did to make it work? Thank you for your cooperation.

DeveloperMarius, 2024-07-03:
Hey,
@virtualabs I also buyed a BBC Micro with version 2.21. Are there any updates on the fix?
Thank you for this awesome project!

~ Marius

TechTune-1, 2024-07-04:
> Hey, @virtualabs I also buyed a BBC Micro with version 2.21. Are there any updates on the fix? Thank you for this awesome project!
> 
> ~ Marius

Hi Marius,
It appears that the MicroBit V2.21 now works with the project, but note that it is only compatible with Bluetooth **Low Energy** connections.