virtualabs/btlejack #85: Does the btlejack-fw-v2.hex working on NRF52840?

DRCRecoveryData, 2023-07-02:
Hi,

I wanna know the btlejack-fw-v2.hex working on NRF52840 dongle?

browntownington, 2023-08-03:
Haha I just came here to ask this exact same question.. specifically it it works with NRF52840.

Did you try it out?

DRCRecoveryData, 2023-08-03:
> Haha I just came here to ask this exact same question.. specifically it it works with NRF52840.
> 
> 
> 
> Did you try it out?

Hi,

I have flash but partition out of memory maybe waitting for developer

browntownington, 2023-08-03:
I just found this closed issue from 2019 saying this chip isn't supported.. so I think we're out of luck.

https://github.com/virtualabs/btlejack/issues/32

bojanpotocnik, 2024-02-02:
The https://github.com/virtualabs/btlejack/issues/73#issuecomment-1325047772 from @virtualabs gave me some hope 
> The whole point was to check if btlejack -i correctly deploys firmware v2 (**designed for nRF52840**) on Micro:Bit v2.21 :D.

However, reading all comments in that issue did not give definitive answer whether all features are now working on nRF52 (nRF52832, nRF52833 and nRF52840 seems to be very similar from the radio perspective) or not.

nosed07, 2024-03-13:
Hello there,

Do we have an update on this? is the nrf52840 supported? have anyone tried?